jsdelivr/globalping-hwprobe #12: restarts and connection losses

skapytek, 2023-01-26:
Hi,
i got my globalping device yesterday and installed it today.
Sometimes i can login with "ssh [logs@192.xxx.xxx.xxx](mailto:logs@192.xxx.xxx.xxx)
After some time im getting this error message:
Could not chdir to home directory /home/logs: No such file or directory
Error: No such container: globalping-probe

Also not shown on https://api.globalping.io/demo/

It needs some minutes to let me login again and shows up on the api website.

This is the "normal" output when its shown on https://api.globalping.io/demo/
Could not chdir to home directory /home/logs: No such file or directory
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Checking for the latest version
Current version 0.10.1
Latest version 0.11.0
Start self-update process
Self-update finished
[2023-01-25 23:17:25] [INFO] [695] [general] Start probe version 0.11.0 in a production mode
[2023-01-25 23:17:26] [DEBUG] [695] [general] connection to API established

jimaek, 2023-01-26:
Hey, the restarts are normal and they are there for security and stability reasons. The SSH connection is only there for rarely debugging, so no need to monitor it constantly.

closed 2023-01-26 by jimaek

skapytek, 2023-01-27:
Is it possible to change that behaviour?

jimaek, 2023-01-27:
No behaviour can be changed because the probe is read-only. So there is no way to persist any settings and the reboot will purge anything you might do. I'll reply about reboots in the other issue.