jsdelivr/globalping-hwprobe #13: restarts every 30 min or less

skapytek, 2023-01-27:
Hey, the restarts are normal and they are there for security and stability reasons. The SSH connection is only there for rarely debugging, so no need to monitor it constantly.

_Originally posted by @jimaek in https://github.com/jsdelivr/globalping-hwprobe/issues/12#issuecomment-1404987479_

Is it possible to change that behaviour through settings?
Is it normal that the restarts happening every 30 minutes, this arent the weekly updates/restarts.

jimaek, 2023-01-27:
Is it possible to get a full log output for 2 hours for example? To see the reboots. And please send any other additional data or information you might have

skapytek, 2023-01-27:
Could not chdir to home directory /home/logs: No such file or directory
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Checking for the latest version
Current version 0.10.1
Latest version 0.11.0
Start self-update process
Self-update finished
[2023-01-27 16:26:03] [INFO] [696] [general] Start probe version 0.11.0 in a production mode
[2023-01-27 16:26:04] [DEBUG] [696] [general] connection to API established
[2023-01-27 16:26:04] [INFO] [696] [api:connect] connected from (Laakirchen, AT, EU) (lat: 47.9819 long: 13.8217)

nothing more in the logs

skapytek, 2023-01-27:
Now it restarted again:

Could not chdir to home directory /home/logs: No such file or directory
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Checking for the latest version
Current version 0.10.1
Latest version 0.11.0
Start self-update process
Self-update finished
[2023-01-27 22:36:02] [INFO] [696] [general] Start probe version 0.11.0 in a production mode
[2023-01-27 22:36:03] [DEBUG] [696] [general] connection to API established
[2023-01-27 22:36:03] [INFO] [696] [api:connect] connected from (XXXXXXXXXX)

jimaek, 2023-01-29:
Hey, are you sure its a reboot and not an SSH timeout? Does it connect immediately? Because if it was a reboot it would take 2-3 minutes to boot and before it allowed you to connect via SSH.

Have you tried using something like this to connect? 

```
ssh -o ServerAliveInterval=20 IP-Address
```

skapytek, 2023-01-29:
Could not chdir to home directory /home/logs: No such file or directory
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Checking for the latest version
Current version 0.10.1
Latest version 0.11.0
Start self-update process
Self-update finished
[2023-01-27 22:36:02] [INFO] [696] [general] Start probe version 0.11.0 in a production mode
[2023-01-27 22:36:03] [DEBUG] [696] [general] connection to API established
[2023-01-27 22:36:03] [INFO] [696] [api:connect] connected from (XXXXXXXXXXXXXXXX
[2023-01-29 10:04:40] [DEBUG] [696] [general] disconnected from API: (transport close)
[2023-01-29 10:04:42] [DEBUG] [696] [general] connection to API established
[2023-01-29 10:04:42] [INFO] [696] [api:connect] connected from (XXXXXXXXXXXXXXXX
[2023-01-29 10:05:07] [DEBUG] [696] [general] disconnected from API: (transport close)
[2023-01-29 10:05:08] [DEBUG] [696] [general] connection to API established
[2023-01-29 10:05:08] [INFO] [696] [api:connect] connected from (XXXXXXXXXXXXXXXX
[2023-01-29 10:05:43] [DEBUG] [696] [general] disconnected from API: (io server disconnect)
[2023-01-29 10:05:43] [DEBUG] [696] [general] connection to API established
[2023-01-29 10:05:43] [INFO] [696] [api:connect] connected from (XXXXXXXXXXXXXXXX

jimaek, 2023-01-29:
I don't see anything indicating reboots in the last log. You do have API disconnects which shouldn't happen but it's unrelated to this issue

skapytek, 2023-01-29:
When i get disconnected and im trying to reconnect it says "no container named globalping found" and hostkey changes everytime

jimaek, 2023-01-30:
Can you confirm you're using the included power supply connected to the wall? Providing less power that required would result in reboots.

skapytek, 2023-01-30:
Yes its the original power supply. I tested it and it delivers the voltage and ampere it should deliver.

jimaek, 2023-01-31:
Would you be able to flash the SD card with the latest firmware and see if that helps?

You can download the firmware here https://github.com/jsdelivr/globalping-hwprobe/releases/tag/v1.2

Its simple https://github.com/jsdelivr/globalping-hwprobe#flashing-the-sd-card

skapytek, 2023-02-03:
Flashed the latest firmware.

After 4 hours another error.

Could not chdir to home directory /home/logs: No such file or directory
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Checking for the latest version
Current version 0.11.0
Latest version 0.11.0
[2023-02-03 17:46:57] [INFO] [688] [general] Start probe version 0.11.0 in a production mode
[2023-02-03 17:46:58] [DEBUG] [688] [general] connection to API established
[2023-02-03 17:46:58] [INFO] [688] [api:connect] connected from (Laakirchen, AT, EU) (lat: 47.9819 long: 13.8217)
client_loop: send disconnect: Connection reset

skapytek, 2023-02-04:
Since 19 hours its working without any problems so far.

jimaek, 2023-02-04:
The reboots are supposed to be happening and they are random. Thats when the hostkey changes. So thats normal as long as it doesnt happen every 30 minutes. 
Looks like it works now :)

closed 2023-02-04 by jimaek