jsdelivr/globalping-hwprobe #20: 7 packets transmitted, 6 received

skapytek, 2023-06-10:
from every other pc there are no packets dropped

```
[2023-06-10 13:37:48] [WARN] [696] [status-manager] ping test result don't match criterias:
{
  status: 'finished',
  resolvedAddress: '193.0.14.129',
  resolvedHostname: 'k.root-servers.net',
  timings: [
    { ttl: 57, rtt: 27 },
    { ttl: 57, rtt: 16.7 },
    { ttl: 57, rtt: 15.2 },
    { ttl: 57, rtt: 19.6 },
    { ttl: 57, rtt: 15.3 },
    { ttl: 57, rtt: 16.5 }
  ],
  stats: {
    min: 15.215,
    avg: 18.392,
    max: 26.984,
    total: 7,
    loss: 14.2857,
    rcv: 6,
    drop: 1
  },
  rawOutput: 'PING k.root-servers.net (193.0.14.129) 56(84) bytes of data.\n' +
    '64 bytes from k.root-servers.net (193.0.14.129): icmp_seq=1 ttl=57 time=27.0 ms\n' +
    '64 bytes from k.root-servers.net (193.0.14.129): icmp_seq=3 ttl=57 time=16.7 ms\n' +
    '64 bytes from k.root-servers.net (193.0.14.129): icmp_seq=4 ttl=57 time=15.2 ms\n' +
    '64 bytes from k.root-servers.net (193.0.14.129): icmp_seq=5 ttl=57 time=19.6 ms\n' +
    '64 bytes from k.root-servers.net (193.0.14.129): icmp_seq=6 ttl=57 time=15.3 ms\n' +
    '64 bytes from k.root-servers.net (193.0.14.129): icmp_seq=7 ttl=57 time=16.5 ms\n' +
    '\n' +
    '--- k.root-servers.net ping statistics ---\n' +
    '7 packets transmitted, 6 received, 14.2857% packet loss, time 1213ms\n' +
    'rtt min/avg/max/mdev = 15.215/18.392/26.984/4.101 ms'
}
```

jimaek, 2023-06-10:
We use a shorter timeouts so thats probably why. But you can ignore it, we require for at least 2 tests to fail before removing a probe

skapytek, 2023-06-10:
Yeah 2 tests in a row where one packet is always dropped. So probe is removed.

jimaek, 2023-06-13:
Additionally we released an update that changes the endpoints being tested, maybe now your probe passes the tests?

closed 2023-06-26 by jimaek