jsdelivr/globalping-hwprobe #21: Probe not visible on the webapp

Albonycal, 2023-06-12:
Error logs: https://paste.debian.net/1282752/

jimaek, 2023-06-12:
Can you try running these commands on some linux machine a few times and tell me if they fail or not?

```
ping -4 -c 6 -i 0.2 -w 15 k.root-servers.net
ping -4 -c 6 -i 0.2 -w 15 j.root-servers.net
ping -4 -c 6 -i 0.2 -w 15 l.root-servers.net
```

Additionally you can try some other root servers from here https://root-servers.org/ and tell me if some other is more stable for you

Albonycal, 2023-06-12:
[image]
[image]
[image]
The l root server is dropping packets

jimaek, 2023-06-12:
In your logs all of them seem to be doing it, so you need to run these commands a few times to be sure.

Also try this and let me know if it improves things or not:

```
ping -4 -c 6 -i 0.3 -w 15 l.root-servers.net
```

Albonycal, 2023-06-12:
> ping -4 -c 6 -i 0.3 -w 15 l.root-servers.net
Still dropping packets, other ones are not (tested multiple times)

Albonycal, 2023-06-12:
> In your logs all of them seem to be doing it, 

No they aren't, only `l` rootserver is dropping packets.

a root server is fine, no packetloss

jimaek, 2023-06-12:
Please send the latest logs from your probe

Albonycal, 2023-06-13:
https://paste.debian.net/1282818/

jimaek, 2023-06-13:
I pushed another update, check if your probe now passes the test

Albonycal, 2023-06-14:
still fails

jimaek, 2023-06-14:
So far it seems that your internet is indeed unstable. I even included an endpoint by an Indian provider.

The only other potential problem I see is the "-i 0.2 " parameter. So please run many pings over the period of an hour to the failing endpoints in your logs using different values of that parameter to compare if there is a difference. e.g. 0.2, 0.3, 0.4, 0.5

Albonycal, 2023-06-14:
I don't think my network is unstable.

Albonycal, 2023-06-14:
Oh wait, I restarted the probe and it downloaded the latest update. And the probe is up now

Albonycal, 2023-06-14:
Also, I think using geoip database for the location of the probe is not a good idea, there should be a way to hardcode the city. It's not correct. 
[image]

jimaek, 2023-06-14:
> Oh wait, I restarted the probe and it downloaded the latest update. And the probe is up now

Great!

>Also, I think using geoip database for the location of the probe is not a good idea, there should be a way to hardcode the city. It's not correct.

You will get this option once our dashboard goes live. You will also get stats and other features

closed 2023-06-14 by jimaek