jsdelivr/globalping-hwprobe #22: Quality control ping test result is unsuccessful

skapytek, 2023-07-21:
_[2023-07-21 14:10:31] [DEBUG] [general] 'http' request 3Yx2QcDSKzwFKhDz received.
[2023-07-21 16:31:41] [WARN] [status-manager] Quality control ping test result is unsuccessful: k.root-servers.net 14.2857% packet loss. Test pass.
[2023-07-21 17:07:09] [DEBUG] [general] 'http' request WMVQfaScKvz0dFVP received.
[2023-07-21 17:31:51] [WARN] [status-manager] Quality control ping test result is unsuccessful: ns1.registry.in 14.2857% packet loss. Test pass.
[2023-07-21 18:52:04] [WARN] [status-manager] Quality control ping test result is unsuccessful: ns1.registry.in 14.2857% packet loss. Test pass.
[2023-07-21 19:03:26] [DEBUG] [general] 'ping' request xCRY5PDyw1FRLfDI received.
[2023-07-21 19:32:16] [WARN] [status-manager] Quality control ping test result is unsuccessful: ns1.registry.in 14.2857% packet loss. Test pass._

Strange cause other devices in the same network dont have any problems pinging the domains

jimaek, 2023-07-22:
As you can see the probe passes the tests "Test pass.". Its just a warning but we require more packet loss to disable a probe. So it should not impact you :)

closed 2023-07-22 by jimaek

skapytek, 2023-08-02:
[2023-08-02 16:17:11] [WARN] [status-manager] Quality control ping test result is unsuccessful: ns1.dns.nl 14.2857% packet loss. Test pass.

from the same network and from different devices (not filtering activated) all ping commands run without any problem:
[2023-08-02 16:17:11] [WARN] [status-manager] Quality control ping test result is unsuccessful: ns1.dns.nl 14.2857% packet loss. Test pass.
So its an hwprobe issue