jsdelivr/globalping-hwprobe #24: errors after reboot

skapytek, 2023-08-09:
This is the output from the logs:

```
      _closeFrameSent: false,
      _closeMessage: <Buffer >,
      _closeTimer: null,
      _extensions: {},
      _protocol: '',
      _readyState: 2,
      _receiver: null,
      _sender: null,
      _socket: null,
      _bufferedAmount: 0,
      _isServer: false,
      _redirects: 0,
      _url: 'ws://api.globalping.io/socket.io/?version=0.14.0&EIO=4&transport=websocket',
      _req: null,
      [Symbol(kCapture)]: false
    },
    [Symbol(kType)]: 'error',
    [Symbol(kError)]: Error: getaddrinfo ENOTFOUND api.globalping.io
        at GetAddrInfoReqWrap.onlookup [as oncomplete] (node:dns:107:26) {
      errno: -3007,
      code: 'ENOTFOUND',
      syscall: 'getaddrinfo',
      hostname: 'api.globalping.io'
    },
    [Symbol(kMessage)]: 'getaddrinfo ENOTFOUND api.globalping.io'
  },
  context: undefined,
  type: 'TransportError'
}
Error: websocket error
    at WS.onError (file:///app/node_modules/engine.io-client/build/esm-debug/transport.js:40:37)
    at ws.onerror (file:///app/node_modules/engine.io-client/build/esm-debug/transports/websocket.js:82:37)
    at WebSocket.onError (/app/node_modules/ws/lib/event-target.js:220:18)
    at WebSocket.emit (node:events:513:28)
    at ClientRequest.<anonymous> (/app/node_modules/ws/lib/websocket.js:728:15)
    at ClientRequest.emit (node:events:513:28)
    at Socket.socketErrorListener (node:_http_client:496:9)
    at Socket.emit (node:events:513:28)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
[2021-11-20 19:39:29] [ERROR] [687] [general] connection to API failed: websocket error
{
  description: ErrorEvent {
    [Symbol(kTarget)]: WebSocket {
      _events: [Object: null prototype],
      _eventsCount: 4,
      _maxListeners: undefined,
      _binaryType: 'nodebuffer',
      _closeCode: 1006,
      _closeFrameReceived: false,
      _closeFrameSent: false,
      _closeMessage: <Buffer >,
      _closeTimer: null,
      _extensions: {},
      _protocol: '',
      _readyState: 2,
      _receiver: null,
      _sender: null,
      _socket: null,
      _bufferedAmount: 0,
      _isServer: false,
      _redirects: 0,
      _url: 'ws://api.globalping.io/socket.io/?version=0.14.0&EIO=4&transport=websocket',
      _req: null,
      [Symbol(kCapture)]: false
    },
    [Symbol(kType)]: 'error',
    [Symbol(kError)]: Error: getaddrinfo ENOTFOUND api.globalping.io
        at GetAddrInfoReqWrap.onlookup [as oncomplete] (node:dns:107:26) {
      errno: -3007,
      code: 'ENOTFOUND',
      syscall: 'getaddrinfo',
      hostname: 'api.globalping.io'
    },
    [Symbol(kMessage)]: 'getaddrinfo ENOTFOUND api.globalping.io'
  },
  context: undefined,
  type: 'TransportError'
}
Error: websocket error
    at WS.onError (file:///app/node_modules/engine.io-client/build/esm-debug/transport.js:40:37)
    at ws.onerror (file:///app/node_modules/engine.io-client/build/esm-debug/transports/websocket.js:82:37)
    at WebSocket.onError (/app/node_modules/ws/lib/event-target.js:220:18)
    at WebSocket.emit (node:events:513:28)
    at ClientRequest.<anonymous> (/app/node_modules/ws/lib/websocket.js:728:15)
    at ClientRequest.emit (node:events:513:28)
    at Socket.socketErrorListener (node:_http_client:496:9)
    at Socket.emit (node:events:513:28)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)

```

jimaek, 2023-08-09:
If you go to api.globalping.io on your computer does it open correctly?

skapytek, 2023-08-12:
[image]

jimaek, 2023-08-12:
Ok then its not a DNS issue or something on our side. You need to ensure that your DHCP server gives out valid DNS servers.
Thats the DNS server that the probe is using and that's what fails to resolve the domain.

closed 2023-08-12 by jimaek

skapytek, 2023-08-14:
What dns server is the probe using?

jimaek, 2023-08-14:
The one you provide via DHCP. Check your router's settings

jimaek, 2023-08-14:
I feel like your backup is the issue, root servers are not resolvers. Maybe replace it with 8.8.8.8 from Google

skapytek, 2023-08-14:
the k is through my own dns ive setup in my network as backup its also resolving for all other clients but i changed it for the hwprobe.