jsdelivr/globalping-hwprobe #25: ns1.registry.in 14.2857% packet loss

skapytek, 2023-08-20:
Ive tested it with a new pc and setup only hwprobe docker.
DNS same as all other devices and ive tested it with different dns (1.1.1.1 and 8.8.8.8).
New pc is working flawlessy and no dns package losses.
Probe still produces package losses:
[2023-08-20 09:14:19] [WARN] [status-manager] Quality control ping test result is unsuccessful: ns1.registry.in 14.2857% packet loss. Test pass.
[2023-08-20 09:34:22] [WARN] [status-manager] Quality control ping test result is unsuccessful: ns1.registry.in 14.2857% packet loss. Test pass.

              Its not something I can replicate on my probes, they pass all tests 100% of the time. So I dont think its a hardware issue.
If you want you could run a docker container probe on your PC and compare the results.

But its not even a failure, the log you sent is just a warning, you can see at the end that the probe was disconnected ` Test pass.`

_Originally posted by @jimaek in https://github.com/jsdelivr/globalping-hwprobe/issues/23#issuecomment-1663815981_

jimaek, 2023-08-20:
But as you can see the test still passes, so it has no effect on your probe :) 
We made that logic flexible to ensure there are not such issues.

I can see the total number of probes globally failing the QA test and it's under 1.5%. If the devices had a problem we would see much bigger numbers.

I am not sure how to help you because there is no problem to solve here really, your probe is online and passes the tests and the hardware seems stable based on our tests and the stats.

I would advice to wait until we release our online dashboard that will give you more visibility into what the probe is doing

closed 2023-08-20 by jimaek

skapytek, 2023-08-20:
```

[2023-08-20 18:32:43] [WARN] [status-manager] Quality control ping test result is unsuccessful: ns1.registry.in 14.2857% packet loss.
[2023-08-20 18:32:43] [WARN] [status-manager] Quality control ping test result is unsuccessful: k.root-servers.net 14.2857% packet loss.
[2023-08-20 18:32:43] [WARN] [status-manager] Quality control ping test result is unsuccessful: ns1.dns.nl 14.2857% packet loss.
[2023-08-20 18:32:43] [WARN] [status-manager] Quality control ping tests failed due to bad internet connection. Retrying in 10 minutes. Probe temporarily disconnected.

```

skapytek, 2023-08-20:
now probe got disconnected

alexey-yarmosh, 2023-08-21:
Probe will retry the QA test in 10 mins, if it is successful, probe is connected back.

alexey-yarmosh, 2023-08-21:
`-w 15`, so it is 15 secs