jsdelivr/globalping-hwprobe #26: Probe not working / segfaulting

Daniel15, 2023-10-13:
I wasn't sure where to report this, and I guess here is as good a place as any.

I've had one of the hardware probes since last year. It was working great, but now I don't see it on the Globalping site. When I ssh to `logs@`, it throws a segfault:

```
Could not chdir to home directory /home/logs: No such file or directory
unexpected fault address 0x284bf7c
fatal error: fault
[signal SIGSEGV: segmentation violation code=0x2 addr=0x284bf7c pc=0x284bf7c]

goroutine 1 [running, locked to thread]:
runtime.throw(0x1b9c5c5, 0x5)
        /usr/lib/go/src/runtime/panic.go:1117 +0x60 fp=0x329f63c sp=0x329f628 pc=0x816a78
runtime.sigpanic()
        /usr/lib/go/src/runtime/signal_unix.go:741 +0x1d0 fp=0x329f654 sp=0x329f63c pc=0x8300f8
github.com/docker/cli/vendor/k8s.io/api/apps/v1.init.0()
        /home/compiler/globalping-hwprobe/poky/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/docker-ce/20.10.8-ce+gitd24c6dc5cf5e68dfb30027b2db454099566a9b9e-r0/git/src/import/.gopath/src/github.com/docker/cli/vendor/k8s.io/api/apps/v1/generated.pb.go:835 +0x4ec fp=0x329f658 sp=0x329f658 pc=0x177490c
github.com/docker/cli/vendor/k8s.io/api/apps/v1.init.0()
        /home/compiler/globalping-hwprobe/poky/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/docker-ce/20.10.8-ce+gitd24c6dc5cf5e68dfb30027b2db454099566a9b9e-r0/git/src/import/.gopath/src/github.com/docker/cli/vendor/k8s.io/api/apps/v1/generated.pb.go:835 +0x4ec fp=0x329f658 sp=0x329f658 pc=0x177490c
runtime.doInit(0x2c83490)
        /usr/lib/go/src/runtime/proc.go:6292 +0x50 fp=0x329f784 sp=0x329f658 pc=0x828fec
runtime.doInit(0x2c88ab0)
        /usr/lib/go/src/runtime/proc.go:6292 +0x50 fp=0x329f8b0 sp=0x329f784 pc=0x828fec
runtime.doInit(0x2c85688)
        /usr/lib/go/src/runtime/proc.go:6292 +0x50 fp=0x329f9dc sp=0x329f8b0 pc=0x828fec
runtime.doInit(0x2c88280)
        /usr/lib/go/src/runtime/proc.go:6292 +0x50 fp=0x329fb08 sp=0x329f9dc pc=0x828fec
runtime.doInit(0x2c89d98)
        /usr/lib/go/src/runtime/proc.go:6292 +0x50 fp=0x329fc34 sp=0x329fb08 pc=0x828fec
runtime.doInit(0x2c85420)
        /usr/lib/go/src/runtime/proc.go:6292 +0x50 fp=0x329fd60 sp=0x329fc34 pc=0x828fec
runtime.doInit(0x2c85af8)
        /usr/lib/go/src/runtime/proc.go:6292 +0x50 fp=0x329fe8c sp=0x329fd60 pc=0x828fec
runtime.doInit(0x2c850b0)
        /usr/lib/go/src/runtime/proc.go:6292 +0x50 fp=0x329ffb8 sp=0x329fe8c pc=0x828fec
runtime.main()
        /usr/lib/go/src/runtime/proc.go:208 +0x278 fp=0x329ffe4 sp=0x329ffb8 pc=0x81978c
runtime.goexit()
        /usr/lib/go/src/runtime/asm_arm.s:841 +0x4 fp=0x329ffe4 sp=0x329ffe4 pc=0x84dcc0

goroutine 21 [chan receive]:
github.com/docker/cli/vendor/k8s.io/klog.(*loggingT).flushDaemon(0x2cef958)
        /home/compiler/globalping-hwprobe/poky/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/docker-ce/20.10.8-ce+gitd24c6dc5cf5e68dfb30027b2db454099566a9b9e-r0/git/src/import/.gopath/src/github.com/docker/cli/vendor/k8s.io/klog/klog.go:1010 +0x70
created by github.com/docker/cli/vendor/k8s.io/klog.init.0
        /home/compiler/globalping-hwprobe/poky/build/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/docker-ce/20.10.8-ce+gitd24c6dc5cf5e68dfb30027b2db454099566a9b9e-r0/git/src/import/.gopath/src/github.com/docker/cli/vendor/k8s.io/klog/klog.go:411 +0x13c
Connection to 10.30.2.2 closed.
```

jimaek, 2023-10-14:
Hey, thanks for reporting. Does this happen every single time you connect, even after reboots?

Daniel15, 2023-10-14:
> [Tracking image][https://github.com/notifications/beacon/AAAWOHLUSMYEOVFGGZVTJ6LX7JERVA5CNFSM6AAAAAA57YN54OWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTJCFCTI.gif]

Yeah, the same thing happens after reboots too. Same error message.
-- 
Regards,
Daniel Lo Nigro
https://d.sb[https://d.sb/]

jimaek, 2023-10-14:
Thats the first time we see that error. Do you maybe have an extra SDCard you could flash and test? In case its a hardware issue with the card itself.
Otherwise I think the best would be if you could ship us the probe to take a look and order a new one via https://hubs.ly/Q01-735y0

jimaek, 2023-10-19:
Hey, please let me know if it worked :) I want to take this into account when ordering future hardware to ensure it doesn't happen again

Daniel15, 2023-10-19:
Hey @jimaek, sorry for the delay. I haven't had time to test with a different MicroSD card. I might have a spare one somewhere - I'll see if I can get time this weekend. How do I flash the SD card?

Daniel15, 2023-10-25:
Good news - I reflashed the SD card it came with, and now it seems to be working fine.

Thanks!
Daniel

closed 2023-10-25 by Daniel15

jimaek, 2023-10-25:
Great, could you take a picture of the bad card? I want to avoid buying them in the future

Daniel15, 2023-11-04:
@jimaek Here you go. Sorry for the delay:

[image]

jimaek, 2023-11-04:
Of all the cards I bought this is the one I didn't expect to fail. I noted it. Thank you!
We will soon order more probes and I will avoid this card

risturiz, 2024-03-27:
Can confirm same problem with same card... I flashed sandisk new one and its working again

JabLuszko, 2024-03-28:
also can throw +1 for same problem, same card. (LET batch I guess :D)
Can't tell for how long as I didn't check logs in months.

jimaek, 2024-03-28:
Looks like the SD Card I purchased werent very reliable :( Can you please send me their names so I could blacklist them?
I'm currently testing some Kioxia cards and so far they're very stable, I'll probably use them in the next batch