jsdelivr/globalping-hwprobe #28: Update container

jimaek, 2023-10-20:
We need the firmware to update the container without user interaction as it happens now. We need to find the most optimal way to do so.
Something to consider:
- People dont like slow boots. So pulling the container on boot is probably not the best idea.
- At some point the old container will stop working completely
- SD Cards suck and we need to protect them

Can the firmware cache the docker container on card and then update it on a schedule? e.g. 

1. We compile the firmware with probe version 0.25.0, the device boots and starts the container. 
2. The old pre-baked container will update the code inside itself and start the probe. The container at this point is based on 0.25.0 but runs latest code v0.35.0
3. The probe is online and works
4. A parallel cron/schedule runs `docker pull ghcr.io/jsdelivr/globalping-probe` once per day and permanently writes it to the SD card. Only if there are changes.
5. Scheduled reboot happens after 5 days.
6. The firmware now boots the cached container from last pull, and not the one it came with.

kernelgurumeditation, 2023-10-23:
It will be very difficult to achieve our reliability in the sdcard using the docker pull approach.
The time window of leaving the sdcard fs r/w  is just too big.
Given that the hardware probes are limited in RAM and sdcard access speed, I suggest an unorthodox approach to the docker update.
We could do the docker pull on the server side(github ?), compress everything in a tar.gz file, and create a sha1 checksum file of the tar.gz contents.
The hardware probes would regularly check the sha1 checksum file in the server, and in case it differs, it would just download the tar file from the server and update the bundled container.
The sha1 checksum file would also be useful to detect corruptions and fallback/force the update retry.

MartinKolarik, 2023-10-23:
The regular docker pull process uses checksums as well so I don't see what would be the advantage here.

kernelgurumeditation, 2023-10-23:
> The regular docker pull process uses checksums as well so I don't see what would be the advantage here.

Easy, the advantage is the checksum verification is done in read-only mode, and it's done on one sequential file and not in hundreds. Also, the docker command is complex the opposite of the sha1sum command

jimaek, 2023-10-24:
I want to avoid any extra steps on the server side.
Have you looked into something like this? https://docs.docker.com/engine/reference/commandline/save/ or 

https://github.com/containers/skopeo

jimaek, 2023-12-20:
Can you please send an update regarding this task? I am not sure what the state is here or what we're waiting for :)

kernelgurumeditation, 2023-12-21:
Hey!, well I got it half done, and when I was doing a test cycle the sdcard died :(  and took some of my development work that I was doing directly on the card :rage: :disappointed:

I will resume work on this weekend