jsdelivr/globalping-hwprobe #29: Consider adding a second IPv6 registry

MartinKolarik, 2023-10-20:
If GitHub doesn't add IPv6 support before we ship new probes. Related to jsdelivr/globalping-probe#189.

jimaek, 2024-03-16:
lol https://isgithubipv6.live/

MartinKolarik, 2024-03-16:
Not that I expected anything else.

MartinKolarik, 2024-03-16:
So what do you think? Docker Hub or something else?

jimaek, 2024-03-16:
If we had 2 or more mirrors, how would the system decide which one to use? Do we modify the docker config to use different mirrors? Does it use round robin in case of failure? Does lack of ipv6 count?

jimaek, 2024-03-16:
If you add Docker hub to workflow and it works with multi-arch, it automatically gets mirrored to ECR too. So we could have 3 mirrors in total. But only if it makes sense.
Otherwise I could setup a proxy cache for ghcr

MartinKolarik, 2024-03-16:
I don't think that's necessary at all. We would just switch everything to the new one and keep the old one working for existing devices.

jimaek, 2024-03-16:
Your proposal still means pushing to and maintaining 2 registries forever. We might as well include both of them in the config and allow the devices to failover in case one is down or gets blocked by the gov.

MartinKolarik, 2024-03-16:
Well, it is possible, just doesn't seem too useful. Either way we need to decide on a "primary" that's documented everywhere and that must support both protocols, and should be the first one we try here as well. Then we can possibly try others on failure.

jimaek, 2024-03-16:
Regardless of the new registry, ghcr.io, is already documented and used everywhere and will need to continue to work forever. So that doesn't change.

The question is how do we add ipv6 support and ideally fix China https://github.com/jsdelivr/globalping-probe/issues/198 in a single change.

Adding a list of mirrors to the docker config sounds like the best solution. Now if the mirrors are real registries or just proxy caches thats a separate issue.

MartinKolarik, 2024-03-16:
As I see it, we need:

1. An address that supports both IPv4 and IPv6 that will be the new primary source and will be used in all updated docs. ghcr.io, of course, remains and can even be used here as a fallback, but shouldn't be the one we advertise on the website because of IPv6.
2. An address that works better in China.

Now, if 1) and 2) are the same, excellent. If not, 2 can be used after some geo detection automatically in HW probes, and be documented as an additional option for people using docker probes.

Also, if 1) ends up being a proxy for ghcr.io that just adds IPv6, I suppose that's fine. What matters is that all our docs show something that works in all environments.

jimaek, 2024-03-17:
We could create a custom domain like registry.globalping.io and proxy cache ghcr via Cloudflare workers https://github.com/ciiiii/cloudflare-docker-proxy

This way no scripts change, everything is still pushed to ghcr and all we have to do is change the docs. And we still control the domain so in the future we could change the underlying system if needed.

MartinKolarik, 2024-03-17:
If that works reliably, sure, but would need some research. I don't know enough about the registry APIs to say if it's a good idea to use a simple proxy like this, plus we'd need to add some restrictions to make only our projects available.

jimaek, 2024-03-18:
If we want a custom domain with CF workers we need to host the domain with CF. globalping.io is not with CF so cant use it. 

So we either decide to use a different domain with CF or consider duplicating all scripts and logic to also push to docker hub. 
By pushing to Hub we also get the option to use Amazon's free proxy https://gallery.ecr.aws/. We then make primary the one that performs best in all regions

MartinKolarik, 2024-03-18:
Overall duplicating the scripts and pushing to docker hub seems easier to me.

MartinKolarik, 2024-03-19:
https://hub.docker.com/repository/docker/globalping/globalping-probe/general

@jimaek do you want to do any perf tests before we move with the changes here or should we go ahead and change the image source?

jimaek, 2024-03-19:
Google is faster for me and many others complain about Hub's speed, but since it's the fastest one in China we should just use it for everyone. 
Maybe document someone the other mirrors too for anyone technical enough to use them

MartinKolarik, 2024-03-19:
All right, so @kernelgurumeditation please replace all uses of the current image with https://hub.docker.com/repository/docker/globalping/globalping-probe/general and verify the builds there work.

kernelgurumeditation, 2024-03-19:
@MartinKolarik ,
ok, I will do it later today, and run tests for one or two days

kernelgurumeditation, 2024-03-19:
@jimaek , @MartinKolarik 

I don't know what you guys did, but the download tool is failing to grab the container, so no builds are possible at this moment..
Did you guys change the type of the image? it seems that now is OCI...

kernelgurumeditation, 2024-03-20:
@MartinKolarik 

the previous version was: vnd.docker.distribution.manifest.v2+json
now is: vnd.oci.image.index.v1+json

The problem is that the tool we use to download the image, can't download this format.

So right now, we can't build new firmware versions, nor any HW probes in the field can be updated by the users.

jimaek, 2024-03-20:
No way to update the tool being used to support the more modern format?

MartinKolarik, 2024-03-20:
@kernelgurumeditation just to make sure I understand the impact - new builds of the FW are failing but this doesn't affect existing live probes in any way, right?

As a short-term solution, I can try downgrading our build process back to the previous version, which will likely bring back the old manifest format. Long-term, we should add support for the new format here.

kernelgurumeditation, 2024-03-20:
@jimaek 
> No way to update the tool being used to support the more modern format?
It will be a big amount of work.

Yesterday, I tried Scopeo, and it failed to generate a usable image.

kernelgurumeditation, 2024-03-20:
> @kernelgurumeditation just to make sure I understand the impact - new builds of the FW are failing but this doesn't affect existing live probes in any way, right?
>
> As a short-term solution, I can [...]

@MartinKolarik, right now if a user decides to use the USB update method, it will fail

jimaek, 2024-03-20:
But their docs say "skopeo can work with OCI images as well as the original Docker v2 images."

MartinKolarik, 2024-03-20:
I'll revert, and let's move this to a separate task.

jimaek, 2024-03-20:
Wait, no need to revert anything. Its not impacting the production so we have time to look into it

MartinKolarik, 2024-03-20:
Order and priorities... I want to test and close this one first.

jimaek, 2024-03-20:
If you want to test just build it locally and manually push to your own docker hub repo for tests. No need to make changes to prod and back and forth

MartinKolarik, 2024-03-20:
I want @kernelgurumeditation to be able to finish this task. Reverting the change is not an issue. Having a broken build process is.

jimaek, 2024-03-20:
@kernelgurumeditation can use any container from any registry and org space for tests, prod is not related to this. Once we know a fix we can push it to prod. Prod is working, only a local build process is broken, so local testing and fix is also the solution

jimaek, 2024-03-20:
I believe the fix is to use this command instead. dir: is not supported.

```
skopeo copy docker://globalping/globalping-probe:latest docker-archive:image-contents
//skip tar creation step
docker load < image-contents
```

closed 2024-06-12 by MartinKolarik