jsdelivr/globalping-hwprobe #33: Log/report firmware version

MartinKolarik, 2023-12-20:
It seems we don't log the firmware version at all, so I have no idea which one my probe currently has. We should log that on startup and possibly also expose it to the API, same as #27?

jimaek, 2023-12-20:
Why does it matter what firmware version your probe has? I can't really think of a use-case.

MartinKolarik, 2023-12-20:
It'll matter less after #28 but still, I guess for any kind of issue we would want to know it and it's just weird there's no way to check even with ssh access.

jimaek, 2023-12-20:
I think given the hardware nature of this product our main advice will always be to flash the latest version, regardless of which firmware version the user reports.

Firmware releases are rare and won't affect the probe too much, and majority issues so far seem to be related  to the hardware itself like failing SD cards.

So for now it's not needed for sure, maybe in the future we can look into it again

MartinKolarik, 2024-03-20:
Let's do this and also add a new env `GP_HOST_FIRMWARE=version` where `version` is the FW release number from https://github.com/jsdelivr/globalping-hwprobe/releases.

jimaek, 2024-03-20:
Firmware updates are basically updates to the pre-baked container version. So I dont see the point of doing this

MartinKolarik, 2024-03-20:
https://github.com/jsdelivr/globalping-hwprobe/issues/39 is quite clearly a feature of the firmware, unrelated to the container itself. Same goes for https://github.com/jsdelivr/globalping-hwprobe/issues/28 and maybe others. We need to understand what state the probes out there are in.

jimaek, 2024-03-20:
Old firmwares cant update containers so nothing we release now will affect them. Future firmwares have the feature of fallback to previous working container so if we make a bad release we can push a new container and they will eventually pull them.
The fw version is irrelevant

MartinKolarik, 2024-03-20:
@kernelgurumeditation please add next to the existing vars with the value "2.0" hardcoded and @jimaek will update in the future when making new releases.

closed 2024-06-04 by MartinKolarik