jsdelivr/globalping-hwprobe #34: Accessing non-container logs

MartinKolarik, 2023-12-21:
I got to a state where the probe booted, started a container update, and that update failed. Connecting via SSH to get the logs was not helpful in this case because it only said `Error: No such container: globalping-probe` - I already found what's most likely the problem in this case, but:

1. I can see there are some log messages like `echo "Starting container update process" > /dev/tty4` in the code. Are those logs accessible in any way?
2. If not, we can we collect the logs in one file and make that part of the SSH output?

kernelgurumeditation, 2023-12-21:
I assume you started the container upgrade using the USB key, as mentioned in the instructions.
You should be able to recover the probe following the instruction howto to factory reset it,
If it still doesn't start the container, then it's possible that sdcard is faulty.

Regarding the questions about the dev logs, there is a way to access those logs, but you will need a hardware probe with a serial console cable mod.
To access the internal logs of the system is just a matter of following:
cat /dev/vcsX where X is the number of subsystem logs you want to see.

Regarding the collection of logs in a file, I strongly advise against this, this is not a PC, and having a filesystem being actively written is a very bad idea.

MartinKolarik, 2023-12-21:
I see, I suppose the requirement for a special dev setup is acceptable but I was mostly thinking if we can't store those logs in RAM to some extent, the same way as the docker logs work.

kernelgurumeditation, 2023-12-21:
They are in the RAM, just not in files.
There are multiple log streams (watchdog, system monitor, etc....) , during development is very useful to have real-time status, instead of the plain old "tail -f... " approach 

PS: I know the system is very locked up, apologies, I know it's a pain to dev... but that was one of the requirements of @jimaek ... to make the users happy with security....

MartinKolarik, 2023-12-21:
Well, then, I guess my question is just - would it be hard to pipe whatever we have to the SSH output in addition to the current container logs? If yes, let's close this for now but if not, might be useful.

closed 2024-06-04 by MartinKolarik