jsdelivr/globalping-hwprobe #40: v2 final testing

jimaek, 2024-03-20:
After everything is merged and no other features are planned need to make sure the final firmware is stable:

1. Load it on multiple devices. From old batch and new batch for a few weeks.
2. Build a firmware with 0.28 container and test the container update process
3. Build a firmware with 0.29 container and test the container update process (after 0.30 is released)
4. Monitor closely and check logs to ensure its not crashing or has update loops
5. Ensure the weekly reboots don't break anything
6. Ensure the sd card is protected from abuse
7. Ensure RAM is enough to do all of the above, even if the container increases in size
8. other?
8. Start shipping

kernelgurumeditation, 2024-04-21:
@jimaek 
Current test status:

> 1. Load it on multiple devices. From old batch and new batch for a few weeks.
I only have one HW unit.. and more importantly, don't have the exact sdcard brand/model that will be used...

> 2. Build a firmware with 0.28 container and test the container update process
Done, no issues found

> 3. Build a firmware with 0.29 container and test the container update process (after 0.30 is released)
Partially done, (faked the version) no issues found

> 4. Monitor closely and check logs to ensure its not crashing or has update loops
Looks ok, my HW probe has been running a test that simulates increased time speed (to test weekly reboot and version checks)
No issues were found in about 500-week reboot/version check cycles

> 5. Ensure the weekly reboots don't break anything
See issue 4.

> 6. Ensure the sd card is protected from abuse
No writes are done to the card if no new version is available

> 7. Ensure RAM is enough to do all of the above, even if the container increases in size
The RAM size is limited, I will run a couple of tests to see how much the container could safely increase in size

> 8. Measure time to start the container
not done
> 9. other?
> 10. Start shipping
🔥 🥳 😨

jimaek, 2024-04-22:
>I only have one HW unit.. and more importantly, don't have the exact sdcard brand/model that will be used...

I will send you a unit from the new batch with the new SD card once I get them :)

MartinKolarik, 2024-04-22:
We'll also release a new probe version for testing soon.

MartinKolarik, 2024-05-09:
New probe version released. Can be tested and then we can start merging things here.

MartinKolarik, 2024-06-04:
@kernelgurumeditation please test one more time the upgrade from 0.29.0 to v0.31.1 and if everything looks good, let's merge.

kernelgurumeditation, 2024-06-04:
@MartinKolarik, ok, I will do the test tomorrow and report

kernelgurumeditation, 2024-06-05:
@MartinKolarik 
test done, and everything was successful
Probe firmware v2 looks to be stable

skapytek, 2024-06-07:
How to get the v2 probe?

closed 2024-06-12 by MartinKolarik

jimaek, 2024-06-13:
@skapytek Its now live https://github.com/jsdelivr/globalping-hwprobe/releases/tag/v2.0
Let me know if you encounter any errors