jsdelivr/globalping-hwprobe #43: Docker log spamming

jimaek, 2024-04-30:
How is the current docker logging logic configured? What if the container decides to write 100TB of logs? 
There are options to limit that, e.g. https://docs.docker.com/config/containers/logging/local/
Are we using them?

kernelgurumeditation, 2024-04-30:
@jimaek, no, we aren't; I will make the change... probably limit it to 10 M  

Maybe it would be a good idea to have a specialized docker, one that is controlled and will not have crazy, unexpected changes.

jimaek, 2024-04-30:
yeah lets limit to 10MB. I think it writes to a RAM mounted endpoint, right? So it wont touch the SD card either way.

>specialized docke
Not sure what is that :)

kernelgurumeditation, 2024-04-30:
Yep, all writes are done to RAM

>>Not sure what is that :)

Ops, I meant a container image  :)

MartinKolarik, 2024-04-30:
> Maybe it would be a good idea to have a specialized docker, one that is controlled and will not have crazy, unexpected changes.

Yes, I thought about that as well, maybe in the future... But I guess the impact here is lower than I initially realized since when the RAM fills up, the device just restarts and works normally again, right?

kernelgurumeditation, 2024-04-30:
>> But I guess the impact here is lower than I initially realized since when the RAM fills up, the device just restarts and works normally again, right?

Yep!

jimaek, 2024-05-06:
So should we apply the logs limits? The devices have arrived and I'm ready to start flashing the new firmware before shipping

kernelgurumeditation, 2024-05-06:
I made the change and let it run on my HW probe
 let me arrive home and check if everything is okay
If it is okay, I will push an MR with the change in 2 or 3 hours.

MartinKolarik, 2024-05-06:
> So should we apply the logs limits? The devices have arrived and I'm ready to start flashing the new firmware before shipping

I'll still want to finish up the probe update before that so there is some time :)

closed 2024-06-04 by MartinKolarik