jsdelivr/globalping-hwprobe #47: Questions

Ernst2020, 2024-09-03:
Hi,

Two questions:

1. 
Became a sponsor last week, any idea when the probe will arrrive? Contacted you by email from the main webpage but got no reply.

2. 
The documentation here says that the firmware .gz file is ready for flashing to an SD card. But I guess it needs to be unzipped first?

Thanks,

Ernst

jimaek, 2024-09-03:
Hey, the probe was shipped on Monday. Unfortunately I didn't receive any emails, maybe it went to spam.

The probe will arrive ready to be plugged in, no flashing needed.

But if you're using BalenaEtcher you don't need to unzip it, just point to the file as is :)

Ernst2020, 2024-09-03:
Ok. Thanks. Eager to install it. Currently running a software probe but Docker for Windows has a bug in vpnkit whereby traceroute and mtr don’t work.

Asked about flashing because I also ordered a NanoPi Neo 512m to be installed on my 4G Vodafone backup line.

Thanks, keep up this excellent work!

closed 2024-09-03 by jimaek