jsdelivr/globalping-hwprobe #48: Small persistent storage

jimaek, 2024-10-18:
I think it makes sense to create a new small volume with like 10MB of storage to save there something like adoption info, to avoid losing probes when the IP changes when the probe was offline.

@kernelgurumeditation is there a safe way to do it to avoid killing the SD card? Maybe enabling RAM caching of the files, or some other configuration?

MartinKolarik, 2024-10-18:
We already have this in https://github.com/jsdelivr/globalping-probe/issues/240. For HW probes, I was actually thinking we let the firmware itself generate the UUID, store it, and expose it to the probe as a variable. The probe itself doesn't need to write anything.

jimaek, 2024-10-19:
How is it generated and how reliable is it?

MartinKolarik, 2024-10-19:
I would rather preserve a way to change it if needed. It should survive restarts and container updates, but a full FW flash should get a new ID.

kernelgurumeditation, 2024-10-19:
@jimaek 
Well, it uses the factory-set SN and other unique data on the chip. If the factory doesn't commit an error, it should be pretty unique.

jimaek, 2024-10-19:
I assume to expose it we would need to flash a new firmware? If we could enable it for everyone with a probe update it would be interesting. But if not then up to Martin

kernelgurumeditation, 2024-10-19:
Yes, it requires a new firmware. Add it to the next version bucket list :)

MartinKolarik, 2024-11-18:
@kernelgurumeditation the probes will now read `GP_PROBE_UUID` variable and use it when set. Please add a mechanism to generate a random UUID once, store it, and expose it as that variable. The value SHOULD NOT change between container updates but SHOULD change if the user reflashes the firmware.

kernelgurumeditation, 2024-11-18:
@MartinKolarik Ok, I will do it!

kernelgurumeditation, 2024-12-01:
> @kernelgurumeditation Do you have an ETA for the firmware update? I want to flash some new SD cards soon and would prefer to include the update. But if it could take a while I won't wait

@jimaek , sorry for the delay, i was on leave

https://github.com/jsdelivr/globalping-hwprobe/pull/53

closed 2024-12-03 by MartinKolarik