jsdelivr/globalping-hwprobe #49: HW Probe 100% packet loss

wboevink, 2024-10-23:
I got a HW Probe and connected it to my network, all other devices on my network can successfully ping except the HW Probe. What could be wrong?

Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Checking for the latest version
Current version 0.33.0
Latest version 0.33.1
Start self-update process
Self-update finished
[2024-10-21T11:23:58.945Z] Current node.js version v20.13.0
[2024-10-21T11:23:59.101Z] Wanted node.js version v20.13.0
[2024-10-21 11:24:06 +00:00] [INFO] [general] Hardware probe running firmware version 2.1.
[2024-10-21 11:24:06 +00:00] [INFO] [general] Starting probe version 0.33.1 in a production mode with UUID fab56eee-3993-4d53-bcb3-b75488e059e2.
[2024-10-21 11:24:10 +00:00] [DEBUG] [general] Connection to API established.
[2024-10-21 11:24:11 +00:00] [INFO] [api:connect:alt-ips-handler] IP address of the probe: 88.159.61.194
[2024-10-21 11:24:11 +00:00] [INFO] [api:connect:location] Connected from Hengelo, NL, EU (KPN B.V., ASN: 1136, lat: 52.27 long: 6.79).
[2024-10-21 11:24:11 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.registry.in: 100% packet loss.
[2024-10-21 11:24:11 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for k.root-servers.net: 100% packet loss.
[2024-10-21 11:24:11 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.dns.nl: 100% packet loss.
[2024-10-21 11:24:11 +00:00] [WARN] [status-manager] IPv4 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv4.
[2024-10-21 11:24:11 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.registry.in: ping: connect: Network is unreachable.
[2024-10-21 11:24:11 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for k.root-servers.net: ping: connect: Network is unreachable.
[2024-10-21 11:24:11 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.dns.nl: ping: connect: Network is unreachable.
[2024-10-21 11:24:11 +00:00] [WARN] [status-manager] IPv6 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv6.
[2024-10-21 11:24:11 +00:00] [WARN] [status-manager] Both ping tests failed due to bad internet connection. Retrying in 10 minutes. Probe temporarily disconnected.
[2024-10-21 11:34:12 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.registry.in: 100% packet loss.
[2024-10-21 11:34:12 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for k.root-servers.net: 100% packet loss.
[2024-10-21 11:34:12 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.dns.nl: 100% packet loss.
[2024-10-21 11:34:12 +00:00] [WARN] [status-manager] IPv4 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv4.
[2024-10-21 11:34:12 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.registry.in: ping: connect: Network is unreachable.
[2024-10-21 11:34:12 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for k.root-servers.net: ping: connect: Network is unreachable.
[2024-10-21 11:34:12 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.dns.nl: ping: connect: Network is unreachable.
[2024-10-21 11:34:12 +00:00] [WARN] [status-manager] IPv6 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv6.
[2024-10-21 11:34:12 +00:00] [WARN] [status-manager] Both ping tests failed due to bad internet connection. Retrying in 10 minutes. Probe temporarily disconnected.
[2024-10-21 11:44:13 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.registry.in: 100% packet loss.
[2024-10-21 11:44:13 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for k.root-servers.net: 100% packet loss.
[2024-10-21 11:44:13 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.dns.nl: 100% packet loss.
[2024-10-21 11:44:13 +00:00] [WARN] [status-manager] IPv4 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv4.
[2024-10-21 11:44:13 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.registry.in: ping: connect: Network is unreachable.
[2024-10-21 11:44:13 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for k.root-servers.net: ping: connect: Network is unreachable.
[2024-10-21 11:44:13 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.dns.nl: ping: connect: Network is unreachable.
[2024-10-21 11:44:13 +00:00] [WARN] [status-manager] IPv6 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv6.
[2024-10-21 11:44:13 +00:00] [WARN] [status-manager] Both ping tests failed due to bad internet connection. Retrying in 10 minutes. Probe temporarily disconnected.
[2024-10-21 11:54:14 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.registry.in: ping: connect: Network is unreachable.
[2024-10-21 11:54:14 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for k.root-servers.net: ping: connect: Network is unreachable.
[2024-10-21 11:54:14 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.dns.nl: ping: connect: Network is unreachable.
[2024-10-21 11:54:14 +00:00] [WARN] [status-manager] IPv6 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv6.
[2024-10-21 11:54:14 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.registry.in: 100% packet loss.
[2024-10-21 11:54:14 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for k.root-servers.net: 100% packet loss.
[2024-10-21 11:54:14 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.dns.nl: 100% packet loss.
[2024-10-21 11:54:14 +00:00] [WARN] [status-manager] IPv4 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv4.
[2024-10-21 11:54:14 +00:00] [WARN] [status-manager] Both ping tests failed due to bad internet connection. Retrying in 10 minutes. Probe temporarily disconnected.
[2024-10-21 12:04:15 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.registry.in: 100% packet loss.
[2024-10-21 12:04:15 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for k.root-servers.net: 100% packet loss.
[2024-10-21 12:04:15 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.dns.nl: 100% packet loss.
[2024-10-21 12:04:15 +00:00] [WARN] [status-manager] IPv4 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv4.
[2024-10-21 12:04:15 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.registry.in: ping: connect: Network is unreachable.
[2024-10-21 12:04:15 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for k.root-servers.net: ping: connect: Network is unreachable.
[2024-10-21 12:04:15 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.dns.nl: ping: connect: Network is unreachable.
[2024-10-21 12:04:15 +00:00] [WARN] [status-manager] IPv6 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv6.
[2024-10-21 12:04:15 +00:00] [WARN] [status-manager] Both ping tests failed due to bad internet connection. Retrying in 10 minutes. Probe temporarily disconnected.
[2024-10-21 12:14:15 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.registry.in: 100% packet loss.
[2024-10-21 12:14:15 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for k.root-servers.net: 100% packet loss.
[2024-10-21 12:14:15 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.dns.nl: 100% packet loss.
[2024-10-21 12:14:15 +00:00] [WARN] [status-manager] IPv4 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv4.
[2024-10-21 12:14:16 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.registry.in: ping: connect: Network is unreachable.
[2024-10-21 12:14:16 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for k.root-servers.net: ping: connect: Network is unreachable.
[2024-10-21 12:14:16 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.dns.nl: ping: connect: Network is unreachable.
[... 7 lines cut ...]
[2024-10-21 12:24:17 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for k.root-servers.net: ping: connect: Network is unreachable.
[2024-10-21 12:24:17 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.dns.nl: ping: connect: Network is unreachable.
[... cut ...]
[2024-10-21 12:44:18 +00:00] [WARN] [status-manager] Both ping tests failed due to bad internet connection. Retrying in 10 minutes. Probe temporarily disconnected.
[2024-10-21 12:54:19 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.registry.in: 100% packet loss.
[2024-10-21 12:54:19 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for k.root-servers.net: 100% packet loss.
[2024-10-21 12:54:19 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.dns.nl: 100% packet loss.

jimaek, 2024-10-23:
Is the probe part of a special config like vlan or firewall rules? If so try adding a different machine to the same config and try running similar tests manually

wboevink, 2024-10-31:
I'm sorry I missed your replies.

The probe isn't part of a special config. When connected to the same router as my laptop (or any other router) the ping tests fail. Pinging form my laptop to the same address succeeds.

jimaek, 2024-10-31:
Can you run these commands on a Linux device connected to the same network and with the same config, like DNS and firewall rules, that the probe would have?

```
ping -c 3  ns1.registry.in
ping -c 3  k.root-servers.net
ping -c 3  google.com
ping -O -c 3 -i 0.5 -w 10 ns1.registry.in
ping -O -c 3 -i 0.5 -w 10 k.root-servers.net
ping -O -c 3 -i 0.5 -w 10 google.com
```

wboevink, 2024-10-31:
It looks like it works for a couple of minutes, if I'm reading the logs correctly and then it stops working

Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Checking for the latest version
Current version 0.33.0
Latest version 0.33.2
Start self-update process
Self-update finished
[2024-10-31T12:49:55.750Z] Current node.js version v20.13.0
[2024-10-31T12:49:55.853Z] Wanted node.js version v20.13.0
[2024-10-31 12:50:03 +00:00] [INFO] [general] Hardware probe running firmware version 2.1.
[2024-10-31 12:50:03 +00:00] [INFO] [general] Starting probe version 0.33.2 in a production mode with UUID 29cac2b6.
[2024-10-31 12:50:09 +00:00] [DEBUG] [general] Connection to API established.
[2024-10-31 12:50:09 +00:00] [INFO] [api:connect:alt-ips-handler] IP address of the probe: 88.159.61.194
[2024-10-31 12:50:09 +00:00] [INFO] [api:connect:location] Connected from Hengelo, NL, EU (KPN B.V., ASN: 1136, lat: 52.27 long: 6.79).
[2024-10-31 12:50:10 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.registry.in: ping: connect: Network is unreachable.
[2024-10-31 12:50:10 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for k.root-servers.net: ping: connect: Network is unreachable.
[2024-10-31 12:50:10 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.dns.nl: ping: connect: Network is unreachable.
[2024-10-31 12:50:10 +00:00] [WARN] [status-manager] IPv6 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv6.
[2024-10-31 12:50:43 +00:00] [DEBUG] [general] 'http' request JGz4REE4uCwjzeJA received.
[2024-10-31 12:51:01 +00:00] [DEBUG] [general] 'dns' request rghYRv6Dt9raeRUc received.
[2024-10-31 12:51:01 +00:00] [DEBUG] [general] 'dns' request AIUPG6LdP8k0GMdk received.
[2024-10-31 12:51:54 +00:00] [DEBUG] [general] 'http' request ughz2LfdGNVH6mUY received.
[2024-10-31 12:52:01 +00:00] [DEBUG] [general] 'dns' request lNc20EW5ZSzpzr46 received.
[2024-10-31 12:53:54 +00:00] [DEBUG] [general] 'http' request yKiWiLhmuaVQi1GJ received.
[2024-10-31 12:54:02 +00:00] [DEBUG] [general] 'dns' request sKmyOGflhce5t06s received.
[2024-10-31 12:54:02 +00:00] [DEBUG] [general] 'dns' request VHHN909E0q76CukU received.
[2024-10-31 12:54:03 +00:00] [DEBUG] [general] 'ping' request WAsTWg2a8uk5UDxF received.
[2024-10-31 12:55:04 +00:00] [DEBUG] [general] 'http' request D1mXCC5WmDf7mreI received.
[2024-10-31 12:56:01 +00:00] [DEBUG] [general] 'dns' request fm6spz3dpWJpb3B6 received.
[2024-10-31 12:57:01 +00:00] [DEBUG] [general] 'dns' request gDY3yDz8sTpAm665 received.
[2024-10-31 12:57:01 +00:00] [DEBUG] [general] 'dns' request lgONoXbFpe7HOUAm received.
[2024-10-31 12:57:01 +00:00] [DEBUG] [general] 'dns' request O5HOiuqnNbuzqtaM received.
[2024-10-31 12:57:54 +00:00] [DEBUG] [general] 'http' request KaAqBEhZPqkHe4Q8 received.
[2024-10-31 12:58:02 +00:00] [DEBUG] [general] 'dns' request 0U36AjVH8gqyyUpO received.
[2024-10-31 13:00:01 +00:00] [DEBUG] [general] 'dns' request MfG2VTXnGJy7uKGq received.
[2024-10-31 13:00:02 +00:00] [DEBUG] [general] 'dns' request r9kcYVDj0Z3cRm9O received.
[2024-10-31 13:00:04 +00:00] [DEBUG] [general] 'http' request XvdH2UlDb5GpXmKt received.
[2024-10-31 13:00:13 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.registry.in: 100% packet loss.
[2024-10-31 13:00:14 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for k.root-servers.net: 100% packet loss.
[2024-10-31 13:00:14 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.dns.nl: 100% packet loss.
[2024-10-31 13:00:14 +00:00] [WARN] [status-manager] IPv4 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv4.
[2024-10-31 13:00:14 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.registry.in: ping: connect: Network is unreachable.
[2024-10-31 13:00:14 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for k.root-servers.net: ping: connect: Network is unreachable.
[2024-10-31 13:00:14 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.dns.nl: ping: connect: Network is unreachable.
[2024-10-31 13:00:14 +00:00] [WARN] [status-manager] IPv6 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv6.
[2024-10-31 13:00:14 +00:00] [WARN] [status-manager] Both ping tests failed due to bad internet connection. Retrying in 10 minutes. Probe temporarily disconnected.
[2024-10-31 13:10:15 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.registry.in: 100% packet loss.
[2024-10-31 13:10:15 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for k.root-servers.net: 100% packet loss.
[2024-10-31 13:10:15 +00:00] [WARN] [status-manager] IPv4 ping test unsuccessful for ns1.dns.nl: 100% packet loss.
[2024-10-31 13:10:15 +00:00] [WARN] [status-manager] IPv4 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv4.
[2024-10-31 13:10:15 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.registry.in: ping: connect: Network is unreachable.
[2024-10-31 13:10:15 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for k.root-servers.net: ping: connect: Network is unreachable.
[2024-10-31 13:10:15 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.dns.nl: ping: connect: Network is unreachable.
[... cut ...]
[2024-10-31 13:20:16 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for k.root-servers.net: ping: connect: Network is unreachable.
[2024-10-31 13:20:16 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.dns.nl: ping: connect: Network is unreachable.
[2024-10-31 13:20:16 +00:00] [WARN] [status-manager] IPv6 ping tests failed. Retrying in 10 minutes. Probe marked as not supporting IPv6.
[2024-10-31 13:20:16 +00:00] [WARN] [status-manager] Both ping tests failed due to bad internet connection. Retrying in 10 minutes. Probe temporarily disconnected.

wboevink, 2024-10-31:
> Can you run these commands on a Linux device connected to the same network and with the same config, like DNS and firewall rules, that the probe would have?
>
> ```
> ping -c 3  ns1.registry.in
> ping -c [...]

wouter@RW-022084-WRBK:~$ ping -c 3  ns1.registry.in
PING ns1.registry.in (37.209.192.12) 56(84) bytes of data.
64 bytes from ns1.registry.in (37.209.192.12): icmp_seq=1 ttl=55 time=7.04 ms
64 bytes from ns1.registry.in (37.209.192.12): icmp_seq=2 ttl=55 time=6.38 ms
64 bytes from ns1.registry.in (37.209.192.12): icmp_seq=3 ttl=55 time=6.54 ms

--- ns1.registry.in ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 1998ms
rtt min/avg/max/mdev = 6.380/6.654/7.039/0.280 ms
wouter@RW-022084-WRBK:~$ ping -c 3  k.root-servers.net
PING k.root-servers.net (193.0.14.129) 56(84) bytes of data.
64 bytes from k.root-servers.net (193.0.14.129): icmp_seq=1 ttl=58 time=10.5 ms
64 bytes from k.root-servers.net (193.0.14.129): icmp_seq=2 ttl=58 time=8.15 ms
64 bytes from k.root-servers.net (193.0.14.129): icmp_seq=3 ttl=58 time=8.43 ms

--- k.root-servers.net ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 1998ms
rtt min/avg/max/mdev = 8.146/9.033/10.528/1.063 ms
wouter@RW-022084-WRBK:~$ ping -c 3  google.com
PING google.com (142.251.36.14) 56(84) bytes of data.
64 bytes from ams15s44-in-f14.1e100.net (142.251.36.14): icmp_seq=1 ttl=58 time=6.54 ms
64 bytes from ams15s44-in-f14.1e100.net (142.251.36.14): icmp_seq=2 ttl=58 time=5.08 ms
64 bytes from ams15s44-in-f14.1e100.net (142.251.36.14): icmp_seq=3 ttl=58 time=5.65 ms

--- google.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 1998ms
rtt min/avg/max/mdev = 5.082/5.758/6.543/0.601 ms
wouter@RW-022084-WRBK:~$ ping -O -c 3 -i 0.5 -w 10 ns1.registry.in
PING ns1.registry.in (37.209.192.12) 56(84) bytes of data.
64 bytes from ns1.registry.in (37.209.192.12): icmp_seq=1 ttl=55 time=6.75 ms
64 bytes from ns1.registry.in (37.209.192.12): icmp_seq=2 ttl=55 time=8.02 ms
64 bytes from ns1.registry.in (37.209.192.12): icmp_seq=3 ttl=55 time=6.94 ms

--- ns1.registry.in ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 1000ms
rtt min/avg/max/mdev = 6.750/7.237/8.024/0.561 ms
wouter@RW-022084-WRBK:~$ ping -O -c 3 -i 0.5 -w 10 k.root-servers.net
PING k.root-servers.net (193.0.14.129) 56(84) bytes of data.
64 bytes from k.root-servers.net (193.0.14.129): icmp_seq=1 ttl=58 time=7.76 ms
64 bytes from k.root-servers.net (193.0.14.129): icmp_seq=2 ttl=58 time=7.59 ms
64 bytes from k.root-servers.net (193.0.14.129): icmp_seq=3 ttl=58 time=8.60 ms

--- k.root-servers.net ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 1000ms
rtt min/avg/max/mdev = 7.589/7.981/8.600/0.442 ms
wouter@RW-022084-WRBK:~$ ping -O -c 3 -i 0.5 -w 10 google.com
PING google.com (142.251.36.14) 56(84) bytes of data.
64 bytes from ams15s44-in-f14.1e100.net (142.251.36.14): icmp_seq=1 ttl=58 time=4.79 ms
64 bytes from ams15s44-in-f14.1e100.net (142.251.36.14): icmp_seq=2 ttl=58 time=5.26 ms
64 bytes from ams15s44-in-f14.1e100.net (142.251.36.14): icmp_seq=3 ttl=58 time=5.43 ms

--- google.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 1000ms
rtt min/avg/max/mdev = 4.792/5.161/5.433/0.270 ms

wboevink, 2024-10-31:
I connected the device directly to the ISP modem/router and it seems to be working for the last 25 minutes

Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Checking for the latest version
Current version 0.33.0
Latest version 0.33.2
Start self-update process
Self-update finished
[2024-10-31T16:02:43.480Z] Current node.js version v20.13.0
[2024-10-31T16:02:43.585Z] Wanted node.js version v20.13.0
[2024-10-31 16:02:50 +00:00] [INFO] [general] Hardware probe running firmware version 2.1.
[2024-10-31 16:02:51 +00:00] [INFO] [general] Starting probe version 0.33.2 in a production mode with UUID 4be87442.
[2024-10-31 16:02:56 +00:00] [DEBUG] [general] Connection to API established.
[2024-10-31 16:02:56 +00:00] [INFO] [api:connect:location] Connected from Hengelo, NL, EU (KPN B.V., ASN: 1136, lat: 52.27 long: 6.79).
[2024-10-31 16:03:00 +00:00] [INFO] [api:connect:alt-ips-handler] IP address of the probe: 2a02:a441:288e:0:b8b3:37ff:fe8b:1574
[2024-10-31 16:03:01 +00:00] [DEBUG] [general] 'dns' request VsG8VPKBgAfwX04h received.
[2024-10-31 16:03:02 +00:00] [DEBUG] [general] 'mtr' request hDfMnGGoKso3UIzM received.
[2024-10-31 16:03:08 +00:00] [DEBUG] [general] 'http' request DZqWF1cowPEucMA7 received.
[2024-10-31 16:03:14 +00:00] [DEBUG] [general] 'http' request CsxHsGPXY28aOuy7 received.

[2024-10-31 16:04:01 +00:00] [DEBUG] [general] 'dns' request 83NyHDsrgplOquO5 received.
[2024-10-31 16:05:01 +00:00] [DEBUG] [general] 'http' request 0kN1oDZlefAMsbfw received.
[2024-10-31 16:05:02 +00:00] [DEBUG] [general] 'dns' request h6I5gUWGiyQTga2D received.
[2024-10-31 16:05:02 +00:00] [DEBUG] [general] 'http' request nPBnSeOtwUUcdtfx received.
[2024-10-31 16:05:02 +00:00] [DEBUG] [general] 'dns' request 3ubnA7jIuCanXTvG received.
[2024-10-31 16:06:01 +00:00] [DEBUG] [general] 'dns' request CLzOJnBBcqcc2Tiu received.
[2024-10-31 16:06:01 +00:00] [DEBUG] [general] 'dns' request mmtqVahpp4N56fx9 received.
[2024-10-31 16:06:09 +00:00] [DEBUG] [general] 'http' request Q21CT3IljjW6jhDN received.
[2024-10-31 16:06:38 +00:00] [DEBUG] [general] 'http' request QaMM44c86Eod2CXo received.
[2024-10-31 16:07:02 +00:00] [DEBUG] [general] 'dns' request YqNnZxyD4asSrDZr received.
[2024-10-31 16:07:02 +00:00] [DEBUG] [general] 'dns' request rFQ6khbqnZwg2nZg received.
[2024-10-31 16:07:02 +00:00] [DEBUG] [general] 'dns' request pmGYu4fW0xmsBZS8 received.
[2024-10-31 16:07:54 +00:00] [DEBUG] [general] 'http' request mvMpX7nX2SOReadC received.
[2024-10-31 16:09:01 +00:00] [DEBUG] [general] 'dns' request DWMWLbjti5bF5Wlq received.
[2024-10-31 16:09:01 +00:00] [DEBUG] [general] 'dns' request xJ7wWdJ1jhpCZ71d received.
[2024-10-31 16:09:54 +00:00] [DEBUG] [general] 'http' request kGF9zuDUEwnv1cTm received.
[2024-10-31 16:10:01 +00:00] [DEBUG] [general] 'http' request OSQbYIsVFQV33S51 received.
[2024-10-31 16:10:04 +00:00] [DEBUG] [general] 'mtr' request 9nU59csgLLVczLA7 received.
[2024-10-31 16:11:01 +00:00] [DEBUG] [general] 'dns' request qHuYq5UHQqqB9mEP received.
[2024-10-31 16:11:01 +00:00] [DEBUG] [general] 'dns' request 98NtzCMQVmT8RfsK received.
[2024-10-31 16:11:01 +00:00] [DEBUG] [general] 'dns' request OscMJ7Ob2NyK9fwG received.
[2024-10-31 16:12:12 +00:00] [DEBUG] [general] 'http' request MpAbx6WfUFTbLD33 received.
[2024-10-31 16:12:34 +00:00] [DEBUG] [general] 'http' request vQ6WVIEimgAHauvX received.
[2024-10-31 16:13:02 +00:00] [DEBUG] [general] 'dns' request ZDRCAli2aVIRvtIY received.
[2024-10-31 16:13:02 +00:00] [DEBUG] [general] 'dns' request RkNrVIljAvE2wmFk received.
[2024-10-31 16:13:02 +00:00] [DEBUG] [general] 'dns' request BloMQxpGIcnD3iyi received.
[2024-10-31 16:14:01 +00:00] [DEBUG] [general] 'dns' request PqFsBCtoM0xqpKOF received.
[2024-10-31 16:15:02 +00:00] [DEBUG] [general] 'dns' request i4338h8EUDsnyFhk received.
[2024-10-31 16:15:02 +00:00] [DEBUG] [general] 'dns' request FJdgfmZxJcVdoMyY received.
[2024-10-31 16:15:02 +00:00] [DEBUG] [general] 'dns' request Yr0T8K5OVE52BGAh received.
[2024-10-31 16:15:13 +00:00] [DEBUG] [general] 'http' request tlK4mYsYHwifas7b received.
[2024-10-31 16:17:12 +00:00] [DEBUG] [general] 'http' request TXee3B7hZSGl9AIj received.
[2024-10-31 16:17:54 +00:00] [DEBUG] [general] 'http' request uD89OXklOZhFIvWF received.
[2024-10-31 16:18:02 +00:00] [DEBUG] [general] 'dns' request X3mewcRj4Ak50Rsl received.
[2024-10-31 16:19:01 +00:00] [DEBUG] [general] 'dns' request pb0ZfelRQePr2Cao received.
[2024-10-31 16:19:01 +00:00] [DEBUG] [general] 'dns' request Jo9I7p2meCA69uTK received.
[2024-10-31 16:19:01 +00:00] [DEBUG] [general] 'dns' request O6CdqlRDS52HS5np received.
[2024-10-31 16:20:01 +00:00] [DEBUG] [general] 'dns' request hK5QuESRuUCQf0jU received.
[2024-10-31 16:20:01 +00:00] [DEBUG] [general] 'dns' request kjP1vdKn7WQeqS98 received.
[2024-10-31 16:20:03 +00:00] [DEBUG] [general] 'http' request Zf4aW4pHkspFRUsH received.
[2024-10-31 16:21:01 +00:00] [DEBUG] [general] 'dns' request rzdc4NrhHEQNkOeK received.
[2024-10-31 16:21:02 +00:00] [DEBUG] [general] 'dns' request CXULZNZki1zwDgpM received.
[2024-10-31 16:21:16 +00:00] [DEBUG] [general] 'http' request 2j1Azd2lEi5emJE1 received.
[2024-10-31 16:22:08 +00:00] [DEBUG] [general] 'http' request e3EWolA76EOzzzE3 received.
[2024-10-31 16:23:01 +00:00] [DEBUG] [general] 'dns' request G2LrxoRtFkXwqr2P received.
[2024-10-31 16:23:01 +00:00] [DEBUG] [general] 'dns' request nl58KS1eCezhNA20 received.
[... cut ...]
[2024-10-31 16:27:02 +00:00] [DEBUG] [general] 'dns' request 4f43L0EfBVnLXUSp received.
[2024-10-31 16:27:56 +00:00] [DEBUG] [general] 'dns' request BqcnT1NcjTmCJtcP received.
[2024-10-31 16:28:01 +00:00] [DEBUG] [general] 'dns' request 4f7KGBTih1mhm6DO received.
[2024-10-31 16:28:01 +00:00] [DEBUG] [general] 'dns' request v5MlWO30e5cQDGSI received.
[2024-10-31 16:29:01 +00:00] [DEBUG] [general] 'dns' request oFPa1RORfk4ok7Ru received.
[2024-10-31 16:29:06 +00:00] [DEBUG] [general] 'http' request svPevkSkZzvjtgvQ received.

wboevink, 2024-10-31:
So connected directly to the ISP modem/router seems to work.
When connected to the same router/network as my laptop it doesn't work or only for a couple of minutes.
Ping tests from the laptop works. There's no special network config or other firewall rules or other vlan for the HW Probe.

This is my current situation:
ISP 
-> ISP Modem/Router => probe works
-> Router (UDM-SE / Firewall / Gateway) => probe doesn't work
-> Aggregation (USW Aggregation) => doesn't support RJ45
-> Router (USW Enterprise 8 PoE) => probe doesn't work

All outbound requests are allowed through the firewall. The probe is on the same VLAN (1 / default) as my laptop.

I don't understand why it wouldn't work, but I happily leave it connected to the ISP modem/router.

jimaek, 2024-10-31:
I have almost the exact same Ubiquiti setup and I'll try to test some stuff. But at the moment my hw probe works fine connected to the UDM router. Check you also check the ubiquiti logs? The probe's and the general security logs.
Its possible some setting is blocking ICMP requests

jimaek, 2024-11-01:
Make sure this is not enabled https://share.perfstack.net/2024/11/chrome_2024-11-01_12-29-59.png
And of course double check your security and probe logs

closed 2024-11-20 by jimaek