jsdelivr/globalping-hwprobe #50: Probe lives in the past and won't get the correct time.

DutchessNicole, 2024-10-30:


jimaek, 2024-10-30:
Hmm pretty sure we do sync time from pool.ntp.org. https://github.com/jsdelivr/globalping-hwprobe/blob/master/meta-jsdelivr/recipes-jsdelivr/jsdelivr-scripts/files/jsdelivr-firstBoot.sh#L11

Any chance you're blocking it? e.g. a firewall rule on the router

DutchessNicole, 2024-10-31:
Absolutely. There's no reason for devices to look on the internet when there's a perfectly fine NTP server right on the local network.

however with tcpdump I found out that while the probe was requesting a timesync, even with the local NTP server, there was no reply coming. So I had to add the new network to ntpsec. That is something I normally don't have to do since everything already lives on defined networks.

And with that after about 10 minutes the probe suddenly started living in the present.

closed 2024-10-31 by DutchessNicole