jsdelivr/globalping-hwprobe #51: HW probe keeps going down

wiljums, 2024-11-10:
My Globalping HW probe keeps going offline and when I want to view the logs via SSH (Putty) the screen closes almost immediately on several tries.
Then that seems to trigger a kind of restart, because the probe can be reached via SSH a little later. In the log you can see that it starts every time with an update process from version 0.33.0 to 0.33.2 which the log says is finished. Node.js version is v20.13.0. Probe is registered on IPv4 and not reachable on IPv6.
Then it comes back online in the Globalping dashboard, but goes offline again after about 7 minutes and it stays offline until restart.

Example of logs:
Using username "logs".
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Checking for the latest version
Current version 0.33.0
Latest version 0.33.2
Start self-update process
Self-update finished
[2024-11-10T12:46:33.525Z] Current node.js version v20.13.0
[2024-11-10T12:46:33.663Z] Wanted node.js version v20.13.0
[2024-11-10 12:46:40 +00:00] [INFO] [general] Hardware probe running firmware version 2.1.
[2024-11-10 12:46:40 +00:00] [INFO] [general] Starting probe version 0.33.2 in a production mode with UUID xxxxx.
[2024-11-10 12:46:44 +00:00] [DEBUG] [general] Connection to API established.
[2024-11-10 12:46:44 +00:00] [INFO] [api:connect:alt-ips-handler] IP address of the probe: xx.xx.xx.xx
[2024-11-10 12:46:44 +00:00] [INFO] [api:connect:location] Connected from xxx, NL, EU (DELTA Fiber Nederland B.V., ASN: 15435,                                                    lat: xx.xxx long: x.xxx).
[2024-11-10 12:46:46 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.registry.in: ping: connect: Network is un                                                   reachable.
[2024-11-10 12:46:46 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for k.root-servers.net: ping: connect: Network is                                                    unreachable.
[2024-11-10 12:46:46 +00:00] [WARN] [status-manager] IPv6 ping test unsuccessful for ns1.dns.nl: ping: connect: Network is unreach                                                   able.
[2024-11-10 12:46:46 +00:00] [WARN] [status-manager] IPv6 ping tests failed. Retrying in 10 minutes. Probe marked as not supportin                                                   g IPv6.
[2024-11-10 12:46:53 +00:00] [DEBUG] [general] 'dns' request UGFqeYY2fRRq2QDY received.
[2024-11-10 12:47:01 +00:00] [DEBUG] [general] 'dns' request X3OJNe8BfqosGadY received.
[2024-11-10 12:47:21 +00:00] [DEBUG] [general] 'http' request Ul2f5Z0b64e3qZjG received.

jimaek, 2024-11-10:
@kernelgurumeditation Can you please check in case you have ideas? It doesn't sound like a hardware failure but maybe you can suggest some debug steps?

wiljums, 2024-11-11:
Today the probe is not accessible anymore with SSH (Network error: Connection timed out). So it gets from bad to worse.

jimaek, 2024-11-11:
Any chance you have an SD card around to try and flash our firmware? In case it started failing for some reason

wiljums, 2024-11-12:
I do, how do I flash the firmware?

jimaek, 2024-11-13:
Simply download the firmware itself https://github.com/jsdelivr/globalping-hwprobe/releases/download/v2.1/globalping-firmware.gz and then flash that file as is to the sd card using Balena https://etcher.balena.io/

Insert into the device and wait a few minutes, logs should appear as always

wiljums, 2024-12-07:
I have tried multiple sd cards and multiple downloads of the firmware , but the flashing process fails because the file is possibly corrupt. Looks like the SD cards are now bricked, hope that's not the case. Windows says they have to be formatted, but cant....
How to solve this?

jimaek, 2024-12-07:
The file is not corrupt as I tested it. But if the flashing fails it's possible the cards themselves started failing.

You can ignore what Windows says, balena etcher will format the card as part of the flashing process.

Unless you want to use the card as a normal file storage, in that case you need to format it via Windows.

Note that a firmware flashed SD card won't appear in Windows or work like a normal card. Maybe that's what you meant by bricked.

Please try again, and if it fails please send a screenshot.

If everything else fails I can send you a new probe

wiljums, 2024-12-09:
current issue: [https://i.imgur.com/tMd82gd.png](url)

wiljums, 2024-12-09:
The issue: [https://i.imgur.com/OGbCLXR.png](url)

jimaek, 2024-12-09:
Did you unzip it? It says the image is 898MB but the file I uploaded is 231MB. Dont unzip it, just flash it as is

wiljums, 2024-12-09:
I tried Rufus in stead of BalenaEtcher and now it worked. Probe is back online, lets see if it stays online.

wiljums, 2024-12-11:
This morning the probe was offline again. Do you want to see logs?

jimaek, 2024-12-11:
Yes, please provide more details and logs, including what offline means, offline in the dashboard or the device had no light on.

wiljums, 2024-12-11:
Offline means offline in the dashboard. If I ssh to logs@ip-adsress then putty shuts down immediately. Dev logs shows: [https://i.imgur.com/omZafIP.png](url)

wiljums, 2024-12-12:
lights besides the UTP port are one

jimaek, 2024-12-13:
I think it's possible this is a hardware issue. I can send you a new probe to replace it. Just submit this form https://forms.gle/9jSzytJpTDYJmS2W8

wiljums, 2024-12-13:
Form submitted

jimaek, 2024-12-13:
Thank you. I'll send it on Monday but it could take a few weeks to arrive, especially now during holidays.

closed 2024-12-13 by jimaek

MDMCK10, 2024-12-24:
Just thought I'd put this here:

One of my probes (specifically, the older DNSFilter giveaway one) is also having a similar issue, went to check on the dashboard a few days ago and noticed it was marked as offline, trying to ssh to logs@ immediately disconnected, and devlogs@ only said "Container status is NOT running!!".

Reflashed the sd card, re-adopted it, all was well, then today I went to check, and it was offline again, though this time logs@ gave the following message:
```
>ssh logs@192.168.0.88
fatal error: index out of range
runtime: panic before malloc heap initialized

runtime stack:
runtime.throw({0x1ba5d0f, 0x12})
        runtime/panic.go:1198 +0x60 fp=0xbe92bb34 sp=0xbe92bb20 pc=0x7f79d0
runtime.panicCheck1(0x819288, {0x1ba5d0f, 0x12})
        runtime/panic.go:36 +0xb4 fp=0xbe92bb48 sp=0xbe92bb34 pc=0x7f3fd0
runtime.goPanicIndexU(0x800e3d2a, 0x3f79e4)
        runtime/panic.go:93 +0x30 fp=0xbe92bb6c sp=0xbe92bb48 pc=0x7f4134
runtime.moduledataverify1(0x2ca3460)
        runtime/symtab.go:585 +0x51c fp=0xbe92bbcc sp=0xbe92bb6c pc=0x819288
runtime.moduledataverify(...)
        runtime/symtab.go:562
runtime.schedinit()
        runtime/proc.go:687 +0x3c fp=0xbe92bbf8 sp=0xbe92bbcc pc=0x7fbf00
runtime.rt0_go()
        runtime/asm_arm.s:168 +0x84 fp=0xbe92bc38 sp=0xbe92bbf8 pc=0x82bbf4
Connection to 192.168.0.88 closed.
```

devlogs:
```
Every 2.0s: cat /tmp/log_collector                                       globalping-probe-aa87: Tue Dec 24 21:10:58 2024
Tue Dec 24 21:10:57 UTC 2024
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!
Container status is NOT running!!

System WatchDog counter:275 Max:300 LAST_CHANCE:1
System WatchDog counter:276 Max:300 LAST_CHANCE:1
System WatchDog counter:277 Max:300 LAST_CHANCE:1
System WatchDog counter:278 Max:300 LAST_CHANCE:1
System WatchDog counter:279 Max:300 LAST_CHANCE:1
System WatchDog counter:280 Max:300 LAST_CHANCE:1
System WatchDog counter:281 Max:300 LAST_CHANCE:1
System WatchDog counter:282 Max:300 LAST_CHANCE:1
System WatchDog counter:283 Max:300 LAST_CHANCE:1
System WatchDog counter:284 Max:300 LAST_CHANCE:1
System WatchDog counter:285 Max:300 LAST_CHANCE:1
System WatchDog counter:286 Max:300 LAST_CHANCE:1
System WatchDog counter:287 Max:300 LAST_CHANCE:1
System WatchDog counter:288 Max:300 LAST_CHANCE:1
System WatchDog counter:289 Max:300 LAST_CHANCE:1
System WatchDog counter:290 Max:300 LAST_CHANCE:1
System WatchDog counter:291 Max:300 LAST_CHANCE:1
System WatchDog counter:292 Max:300 LAST_CHANCE:1
System WatchDog counter:293 Max:300 LAST_CHANCE:1
System WatchDog counter:294 Max:300 LAST_CHANCE:1
System WatchDog counter:295 Max:300 LAST_CHANCE:1
System WatchDog counter:296 Max:300 LAST_CHANCE:1
System WatchDog counter:297 Max:300 LAST_CHANCE:1
System WatchDog counter:298 Max:300 LAST_CHANCE:1
System WatchDog counter:299 Max:300 LAST_CHANCE:1
System WatchDog counter:300 Max:300 LAST_CHANCE:1
System WatchDog counter:301 Max:300 LAST_CHANCE:1
Container status is faulty
Container recover attempt failed, resorting to full system reboot

Starting JSDELIVR World
Changing compression algo
Change ram disk size
Formating ram disk
Mounting ram disk
Running image/container....
```

@jimaek any ideas?

jimaek, 2024-12-29:
Hey, just FYI, we're looking into this

jimaek, 2025-01-05:
@MDMCK10 Looks like the SD card is failing. Do you have another one to flash and test?

MDMCK10, 2025-01-05:
> @MDMCK10 Looks like the SD card is failing. Do you have another one to flash and test?

@jimaek I had a look around my house and found these three, though I'm not sure if any of them meet the requirements. (I assume whatever the probe may be doing is quite write-heavy, given that it killed the PNY one in about a year).

[image]

Let me know if any of these would work, and if not, which one I should look into getting instead.

jimaek, 2025-01-06:
Actually it's the opposite, its read heavy, nothing is being written really and it runs in RAM. SD Cards just suck in general :(

You can try any of them :)

MDMCK10, 2025-01-08:
> Actually it's the opposite, its read heavy, nothing is being written really and it runs in RAM. SD Cards just suck in general :(
> 
> You can try any of them :)

I actually ended up getting a "Samsung PRO Endurance 32GB" instead and using that, surely since it has endurance in the name it should last more than a year, hopefully.

Either way, flashed and adopted, we'll see if it holds.